bdring/6-Pack_CNC_Controller #7: Modules to test.

bdring, 2020-06-14:
I got a bunch of spindle modules to test.

[image]

The relay one is tested and installed.

[image]

Testing soon. RS485 modbus, 0-10V and 5V PWM.

Bleeptech, 2020-06-26:
Nice! I've been looking for a controller for an XXYZZ build with DM542A external drivers and this looks very promising..
A set of jumpers across the driver sockets should be enough to expose the signals I need.. 
Is there a machine file for this or have I not dug deeply enough into the code?

Bleeptech

P.S. Any chance you'd sell me a partially populated board and a set of the screw connectors?
[ I could breadboard it but they always end up looking like an LA freeway map.. LOL ]

bdring, 2020-06-27:
The latest version is on order. If it checks out I will be selling fully assembled versions.

There is a PCB version of the jumper you are talking about. The jumper just allows you to choose between a common ground or common 5V.

[image]

closed 2020-07-12 by bdring

bdring, 2020-09-22:
[image]

bdring, 2020-09-22:
[image]