bdring/6-Pack_CNC_Controller #8: MPG Jog Pendant Module Proposal

MitchBradley, 2020-06-20:
An input module that could connect to something like this: 

[MPG Pendant](https://www.wish.com/product/5b10e987c25869171fbeb794)

Requires lots of inputs, so probably needs an I2C expander or something like.

KingMo5h, 2020-06-21:
That would be a great if this or similar could be integrated, I wanted to ask this too.
Yes probably needs a expander but the software work will be more challenging i guess.

Bleeptech, 2020-06-26:
It looks like the easiest solution might be adding a CH55x chip to handle the USB host functions.. 

bitluni stuffed one in his ESP32 powered micro game console.. 

https://www.youtube.com/watch?v=po3FBdY0GS4

MitchBradley, 2020-06-27:
It's good to know that is an option - but anyone who has ever dealt with programming USB devices might take issue with the word "easiest".

Bleeptech, 2020-06-27:
Been there.. Done that.. [Good thing I was already nuts before I started..]

ESP32 KE32-WROVER + GY521 with USB HID over Blueooth.. Enumerates as a joystick..

[image]

[Dont ask me to open the can.. Suffice it to say that it looks like R2D2 did a Jackson Pollock..]

yngndrw, 2020-07-17:
The [Contour ShuttleXpress](https://contour-design.co.uk/products/multimedia-controller/shuttlexpress/) might be a good option with a USB Host module.

LinuxCNC has a implementation which would save any reverse engineering:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/user_comps/shuttle.c#L64-L86

jschoch, 2020-07-27:
fixing the jog cancel race condition would be needed first.  I believe a much more complete jogging implementation was planned but it couldn't be fit into OG grbl.  Maybe the underlying jogging implementation should be fixed first?

MitchBradley, 2020-07-27:
Please include references to that race condition.

jschoch, 2020-07-28:
there are quite a few threads on the topic, here are a few.

https://github.com/gnea/grbl/issues/95

https://github.com/gnea/grbl/issues/837

https://github.com/gnea/grbl/issues/734

here is an attempted fix:  grblHAL seems to have this fixed and added a few features to support MPG

https://github.com/terjeio/grbl-Mega