bdring/6-Pack_CNC_Controller #9: LCD Display/Control Module Proposal

MitchBradley, 2020-06-20:
A module to interface to an LCD display, for example:

[Ender LCD Display](https://www.ebay.com/i/254599064474)

It needs 3 input pins for the rotary encoder and its associated pushbutton, 2 output-only pins for the display, and one bidirectional pin for the display.  Optionally, there can be another output pin for the beeper, and another input pin for a pushbutton switch that is conventionally used for system reset.

Lots of software work would be necessary to use something like this.

There are similar displays (see [Reprap Discount Smart Controller](https://www.ebay.com/i/252815565682)) that need more pins - dual IDC cables instead of the one - because they run the display controller in 4-bit-data mode, and they also have an SD card slot.  The external SD card slot does have a mounting advantage, in that the slot can be located in a more convenient location.  That advantage is mitigated be the WebUI ability to upload files to SD from a remote computer.

There are also [touchscreen displays that have an onboard STM chip](https://www.ebay.com/i/223769115207) , that potentially could be used without much software work on the ESP end.  The on-board micro does all the menu stuff, and sends commands back to the gcode controller over an extra serial port.  Unfortunately they send Marlin commands, so it might be necessary to hack their firmware.

KingMo5h, 2020-06-21:
Maybe integrate the lcd of the tft35 or similar into the MPG Jog Pentant Module. 
So everything works with the STM chip on the TFT and it sends only commands to the esp32 like you said but i dont know if the TFT has enough i/o pins or some expander could be included to get everything to work.

Aggebitter, 2020-11-09:
@MitchBradley 
> There are also [touchscreen displays that have an onboard STM chip](https://www.ebay.com/i/223769115207) , that potentially could be used without much software work on the ESP end. The on-board [...]

Firmware for GRBL: [https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC]

MitchBradley, 2020-11-09:
Yeah, I have one of those on one of my 3D printers.  The stock firmware is mildly configurable, but mostly in the way that screens are presented.  There is an open source version of the firmware, so hacking would be possible, if perhaps tedious.

Someone recently informed us of this, which looks much more suitable:  https://github.com/mstrens/grbl_controller_esp32 .  It could be wired to a second serial port on some Grbl_Esp32 pins, especially if one has Bart's new 6-pack board with plenty of spare GPIO pins and plug-in IO modules. It uses a second ESP32, so it ought to be possible to connect it wirelessly by hacking its firmware to talk over bluetooth, which would then work with any Grbl_Esp32 board with no hardware mods.

Skywalkerf34, 2021-12-28:
This is available now on TINDIE here :
 https://www.tindie.com/products/24799/
https://www.tindie.com/products/24731/