bdring/6-Pack_CNC_Controller #11: Version 1.0 Uploaded

bdring, 2020-06-27:
[image]

A new version has been designed, added to [the repo](https://github.com/bdring/6-Pack_CNC_Controller/tree/master/versions/V1p0) and 10 samples ordered.

- Fixed issue where StepStick and SD card conflicted.
- Switched from (2) spindle modules to (5) universal CNC I/O Modules.

The CNC I/O Modules idea is [defined here](https://github.com/bdring/6-Pack_CNC_Controller/wiki/CNC-Module-Definition). More details will follow. 5 different test modules were ordered.

KingMo5h, 2020-06-27:
Looks nice. 
Btw the traces in the left corner (StepStick socket) is different then the others. It has 90° instead of 45° like the others.

bdring, 2020-06-27:
Thanks for the tip on AP64350SP-13