bdring/6-Pack_CNC_Controller #12: Version 1.1 Uploaded

bdring, 2020-07-12:
Testing of version 1.0 went great. There were no real issues with the hardware. I made a few small tweaks and released 1.1 for a small production batch.

- Tented the vias. That was a setting that accidentally got flipped in my software.
- Moved a couple vias away from pads.
- Added a 5V power LED
- A lot of silkscreen improvements.
- Moved the JTAG under CNC Module #1

[image]

Dan018, 2020-07-14:
This board will do everything I am looking for ! Great work!

I would like to get some of these boards made. Is the v1p0 uploaded here OK to produce, or should I wait and produce version 1.1 (v1p1?) when the diptrace is uploaded?

bdring, 2020-07-14:
I am sorry, I forgot to push the changes. V1p1 is available now.

I have 20 fully assembled pieces on order from PCBWay as a production test. They take about 4 weeks. I also have all the modules on order too.

If you want to assemble yourself, you could probably do it faster.