bdring/6-Pack_CNC_Controller #16: 74AHCT595D alternative?

gitxmax, 2020-07-21:
The part seems to be available @ Mouser only, which I can't purchase from. Is there any more common alternative available?
Thanks

bdring, 2020-07-21:
I think this should work.

https://www.digikey.com/product-detail/en/texas-instruments/SN74AHCT595DR/296-26428-2-ND/1567309

BigYuanHead, 2020-08-25:
> I think this should work.
> 
> https://www.digikey.com/product-detail/en/texas-instruments/SN74AHCT595DR/296-26428-2-ND/1567309

No, it can`t. I tried this one, nothing output in Dir and Step pin. I still can not figure out what`s going on.

EsserPrototyping, 2021-04-17:
> > I think this should work.
> > https://www.digikey.com/product-detail/en/texas-instruments/SN74AHCT595DR/296-26428-2-ND/1567309
> 
> No, it can`t. I tried this one, nothing output in Dir and Step pin. I still can not figure out what`s going on.

I ordered the SN74AHCT595D from Farnell:
https://de.farnell.com/texas-instruments/sn74ahct595d/logic-8bit-shift-register-16soic/dp/3120848?ost=3120848

But it´s not working.. because they have send me the WRONG PART! Check the marking on the IC itself.. mine says HC166, that is the SN74HC166 and NOT the SN74AHCT595D.