bdring/6-Pack_CNC_Controller #17: Obsolete Power Supply IC

bdring, 2020-07-21:
The MP1584EN power supply IC obsolete. I have been using it for years and did not notice it until recently. My PCB supplier has no trouble getting them though.

I have a test controller on order with the [Diodes Inc AP64350SP-13](https://www.diodes.com/part/view/AP64350). It is very close in capability and has a higher max input voltage. I will switch to that if it works out.

KingMo5h, 2020-07-22:
Thats the one iam using. Works fine without problems.

abrackers, 2023-01-08:
For the external driver version, the TPS54360 (C44377) is specified, however it seems to have been replaced by the TPS54360B version (C524806). Is this just a drop in replacement or will I need to change some other components?

bdring, 2023-01-09:
Let me Google that for you

I found this.

[image]

Looks like it is just a polymer coating. It should be a drop in replacement.

Please consider donating to the project.
http://wiki.fluidnc.com/#donations