bdring/6-Pack_CNC_Controller #19: Optimize Pin Header Orientation of CNC Modules for Active Cooling

MCTaylor17, 2020-07-25:
This is more of a question than a suggestion because I'm not that familiar with electronics or the vision for the CNC modules.  It also might be a low priority, version 2 or 3 consideration because I suspect it involves a lot of rework.  With that said, let me explain what I'm doing and my observation about the current orientation.

I'm taking a small stab at designing a 3D printed fan enclosure which got me thinking about the thermals.  My high level plan is a top-mounted fan blowing down over the board with dedicated ducting to channel air more directly towards the stepper drivers.  With respect to the CNC modules, I'm thinking I can provide similar ducting for bays 2, 3, and 5 while reserving bays 1 and 2 for lower current applications like sensors.  Based on the universal/prototyping nature of the board, I'm thinking it might be useful to add some butterfly valves to provide selective granularity over which drivers receive the most cooling.

Now, I notice the alignment of the pin headers currently forms a wall from the center, effectively blocking the underside of the boards.  I'm wondering if it would be beneficial to reorient the headers 90 degrees (see green rectangles below) to allow for air flow beneath the boards.  This would provide both the option to allow air flow for active cooling, or to block it off to increase airflow elsewhere.
[image]

This primarily comes down to one question: How likely do you think it is that the underside of the boards will require cooling?  For example, the TMC2208's recommend this type of cooling (see below).  I'm not sure how much of that is a function of space constraints, but "low-profile" CNC modules could be desirable for some applications.  

If you don't envision the underside ever needing cooling, the current orientation is actually quite smart, in which case bravo!!

[image]

bdring, 2020-07-26:
Hey Mike,

Thanks a lot for the input, but it is a little too late to make any changes. The motherboard was prototyped twice, a couple of the modules were prototyped more than once and there is a lot of production hardware no order. I can give a little insight into my thought process on the layout.

### Size

I wanted to appeal to small scale DIY CNC machine designers. I did not want a completed controller to end up being so large it would not be consider for small scale machines. I mocked up a basic set of modules and tried to see  how much width I needed for the external interfaces. The widest was the 4x input module. It has 4 terminal blocks with 3 terminals each.  There also needs to be a mounting screw along that edge. Placing the microcontroller interface on any edge except for the opposite side would have made the modules even wider.

One last thing I considered was a mini, budget, 3 axis version that is about 1/4 the size. This relies on the modules hanging off the motherboard. I have not finished the design, but here is a sneak peek.

[image]

### Isolation

A lot of DIY CNC machines suffer from electrical noise and spikes. This can be difficult to fix and can be a mess of last minute fixes. This is especially true on the inputs. I wanted to design some "bulletproof" modules that you can expect to plug in and just work. Also, when you use relays that are connected to high current DC or AC, you want to isolate that. Putting the external interface as far away from the microcontroller interface seemed like the logical way to solve this.

### Cooling

I did not consider underside cooling. The board currently does not support UART type drivers like the TMC2208. I am not sure why you would choose that driver unless you already had some. You will get better performance out of the TMC2130 and TMC5160 drivers. I assumed top side cooling would be sufficient and I placed 2 fan connectors, one on each end of the board near the drivers. I think if you really want to push some power, you should consider external drivers.

None of the current crop of CNC modules need cooling. If high current modules for motors are used, they should integrate their own cooling. The module socket connector is not designed to provide a lot of current anyway. The spec says that should also come from off board.

I would love to see what you come up with for an enclosure.

MCTaylor17, 2020-07-26:
Thanks for the response. I'm really loving your design philosophy. Sounds very reasonable to me.  Especially the point about isolation considering most of the terminal would sit between it's own and a neighboring header.  Considering you want 3rd party module development, isolating the headers like you have is really smart.  Will save a collective ton of troubleshooting hours over the years 🙇‍♂️

My only question from the above is regarding module cooling.  I was considering ducting for the modules because your video suggested there may be future support for additional drivers.  I'm kinda hearing those will require their own power supply and cooling fan.

For myself, I don't have any plans for high power steppers just yet and part of designing an effective cooling case is to delay that need as long as possible.  I think what I'll do is forget the modules and design strictly for cooling the steppers (and the ESP32 if you think it needs it?), just to keep things simple.  No need to future proof and over engineer a simple enclosure.  After all, I only need 4 or 5 steppers for what I have in mind.  Once I hit 6 I'll start worrying 😂

closed 2020-07-26 by MCTaylor17