bdring/6-Pack_CNC_Controller #20: 10-30 V inductive endstops

raenji-sk, 2020-07-26:
Would it be possible to upgrade the end stop module to support 10-30V inductive sensors (ideally both PNP and NPN)?

bdring, 2020-07-26:
You could build it with a different isolated power supply and led resistors to use 12v NPN sensors.

yngndrw, 2020-07-26:
The [switch input module](https://github.com/bdring/6-Pack_CNC_Controller/wiki/4x-Switch-Input-module) just requires the signal pin to be pulled to ground, so it should already work with open collector / NPN outputs. The power supply for the sensors would need to be handled separately.

PNP sensors would require a little more work, essentially you have to [convert them to an NPN output with an external NPN transistor](https://electronics.stackexchange.com/a/255677):
[image]

KingMo5h, 2020-07-26:
The cheapest isolated DC-DC converter and with 2W I could find are the Meanwell SPUN02L, either 5V, 12V or 15V Output. Input is 5v 12v or 24v. Maybe it helps.

raenji-sk, 2020-07-27:
OK, thanks. I will look into it when I have some time. I've been using GRBL for a long time and am looking for some other hardware as well as the 8 bit is clearly limited these days. This board seems very nice and I would not have to get used to a new controller / software and keep on using GRBL with better hardware.

raenji-sk, 2021-10-08:
@M10CUBE Sorry that was the first and last version. I had the board made in China but never assembled it (I never liked the integrated drivers much to be honest). Went for grblhal and teensy 4.1 board instead. That worked quite well for me for a year or so. But that's history now too as I changed the cnc controller to Uccnc.