bdring/6-Pack_CNC_Controller #21: RC Servo/BESC CNC I/O Module

bdring, 2020-08-02:
Here is a simple CNC I/O Module to add a couple of RC servos or a BESC (Brushless Electronic Speed Controller).

[image]

It has a switch to allow the servos to be powered from the controller's 5V (limited current) or an external power source. Some bigger servos like a higher voltage, so that could be provided through the terminal block. Grbl_ESP32 also supports BESC based  spindles, so you could use this module for that as well.

There are other ways to get more servos via I2C PWM controllers, but this is a just a simple one that requires no changes to the firmware.

bdring, 2020-08-10:
Here it is.

[image]

plangenh, 2020-10-10:
Thanks a lot for this great modular project! 
Are you planning to upload the Source files of this RC Servo Module in the CNC_IO_Modules directory as well?
I am planning on using two servos and would be really interested in the files. I'd order the module on tindie, but shipping to Germany is really expensive.

bdring, 2020-10-10:
I uploaded the source files and gerbers. I am sorry about the shipping cost, but I am only shipping via Priority Mail during the pandemic. The other methods are taking up to 3 months.

Please consider a donation to the Grbl_ESP32 project.

plangenh, 2020-10-11:
Thanks a lot, really appreciated!!
Will definitely donate to the project.