bdring/6-Pack_CNC_Controller #23: Just for fun....3 Pack

bdring, 2020-08-15:
I needed to test a new DC-DC power supply, so I decided to make a mini controller at the same time. 

This is tiny controller that uses CNC I/O Modules. It is about the size of an Arduino Uno. It has sockets for 3 axes and 3 modules. Other than the power supply, there is no active circuitry on the board. It took only about an hour to design. It shows how the modular concept allows you to quickly prototype a controller with professional level accessory support.

[image]

[image]

Skywalkerf34, 2020-08-15:
Nice design, as already said modular design helps a lot, then "just" route a new board and that's it !

bdring, 2020-08-17:
It would be interesting to make a 5V DC-DC module.

It would use the VMot to output 5V using the module pins. That would further reduce the  complexity of the base controller.

closed 2020-09-11 by bdring