bdring/6-Pack_CNC_Controller #24: Getting ready to sell a few

bdring, 2020-08-18:
I am just about ready to sell some controllers. I did a small test production that went perfectly. I have about 15 to sell before I order more. I have all of the modules except the 5V Output (PWM, Lasers Etc). That is due soon. There was a delay on one of the parts.

I want to sell to people who have built a machine successfully before and have some experience with Grbl_ESP32. The goal is to get them into use as fast as possible. The hardware is pretty solid, but the firmware might be tweaked a bit. Expect to have to load the firmware a few times.

If you are interested reply to this post. It would help if you give some details about how you plan to use it and the modules you would need.

The price will be released soon, but expect the main controller to coast in the $40 - $50 range and the modules are $10-$20 range depending on the complexity.

tgirard, 2020-08-19:
I'll buy one. I bought one of your first grbl controllers and loved it. I'm using 2209's with sensorless homing on my MPCNC. I would like to get two relay modules and an input module if you have them. 

Let me know
-Timothy Girard

bdring, 2020-08-19:
Here is the price and availability right now.

$45 for the 6 Pack Motherboard

$20 for the 4x Input Module
$20 for the RS485 Module
$18 for the 0-10V Spindle Module (no stock yet, on order, delivery TBD)
$12 for the Relay Module
$11 for the 5V Output module (no stock yet, on order, delivery soon)
$10 for the RC Servo / BESC module
$4  for external driver adapters.

$25 for a pre-programmed Grbl_ESP32 to your requirements.

They would be sold via PayPal for a while....then Tindie

contact me via bdring  <at> buildlog.net

tgirard, 2020-08-20:
Thanks Bart,

Buildlog.net is not co-operating

 

Can you please just send me a link where I can put the order in and paypal you?

 

 

Thanks 

-Tim

 

contact me via bdring buildlog.net

bdring, 2020-08-20:
Contact me via DM on Slack. [Here is an invite link](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA)

xP373Rx, 2020-08-29:
I was looking for a board like that for a while.. Im running grbl on my cnc with esp32 and i love the simplicity of uploading and working with it. I also own 2  3d printers. Which i would like to use esp32 on. I got marlin compiled for esp32  and its running but esp simply has to few IOs if you have more then 3 axes and extra sensors. You could basically make some mosfet and thermistor modules and make a great wifi printer board out of that. Current solutions work with serial wifi modules which are rly slow on upload. It would be a true board for all cnc and printer nerds ;> As for display , you can use serial and existing marlin compatible displays. Marlin already supports lucs webinterface. the only thing that needs to be added is support for your modules.

Great work on the GRBL port >;

closed 2020-09-11 by bdring