bdring/6-Pack_CNC_Controller #28: wiki updates socket pins and external module soldering tip

jschoch, 2020-08-25:
for the external stepper module  it is easier to solder the inner voltage selection pins first, may be a helpful note to add to those instructions.

for the Socket Pin Number Mapping if the default pins are unlikely to change it would be helpful to add them to the list and for them to match the order on the board.  for example:

```

// #4 GPIO_NUM_34 (input only) default: A_LIMIT_PIN
// #3 GPIO_NUM_35 (input only) default: Z_LIMIT_PIN
// #2 GPIO_NUM_32  default: Y_LIMIT_PIN
// #1 GPIO_NUM_33  default: X_LIMIT_PIN

```

also the i2s_out_xyzabc.h is different, it isn't clear why but it would be nice if it matched or if it was clear what the difference was intended for.

```

// Socket #1
// #1 GPIO_NUM_33  default: C_LIMIT_PIN
// #2 GPIO_NUM_32  default: B_LIMIT_PIN
// #3 GPIO_NUM_35 (input only) default: A_LIMIT_PIN
// #4 GPIO_NUM_34 (input only) default: Z_LIMIT_PIN

```

bdring, 2020-08-25:
I don't follow you on the socket mapping suggestion. 

FYI: On the next version of the 6 Pack (on order), the silkscreen changes to remove any axis references. It is very likely that people will not use the sockets for XYZABC. 

The only hardwired issue is the stallguard wiring, so there is a reference to that. If you are using the stallguard for motor number #1, you cannot use the socket #1 pin attached to it.

[image]

jschoch, 2020-08-26:
`I don't follow you on the socket mapping suggestion.`

with the 4 input board you can't see any of the silk screen so you have to take it out to figure out what is connected.  The order of the pins in the wiki also doesn't match the physical order of pins.  You could name the "signal" pins like this:
Sig<Module #>:<1..4>, e.g Sig5:2 which would map to ISO(27)

regarding i2s_out_xyzabc.h vs i2s_out_xyzabc_trinamic.h I now see that there are also 6-pack specific machine files in the Machines dir, and I had made the bad assumption that i2s_out_xyzabc_trinamic.h was for the 6-pack board.  For someone using external steppers it may not be clear which file to use.

bdring, 2020-08-26:
The numbers are correct as far as I can tell.  Are you orienting signal #1 correctly. With your format Sig5:2 would be ISO(25) not 27.

I can add these two images for reference.

[image]

[image]

bdring, 2020-08-26:
The I2S_out... machine def predates the 6 Pack. It was used for I2S when it was still a bread board.

Most people are looking for 6 Pack in the name. I can kill it off soon.

jschoch, 2020-08-26:
sry, i don't know which diagram I got ISO(27) from.  Sig5:2 would be ISO(25) would be a good way to disambiguate the mappings.  

I also noticed there are no control pins mapped by default, maybe no one uses them.  It would be nice if you had them mapped out as defaults and or commented out defaults in a way that had shows what would need to be swapped.  For example 

```
/*

// To enable control pins you must disable another function such as MIST 
// Instructions for setting up external pullups: <link>
// Instructions for the opto input module control pin configuration<link> 

 #define CONTROL_RESET_PIN       GPIO_NUM_2  // default pin: Sig2:1   Swap for MIST default pin
 #define CONTROL_FEED_HOLD_PIN   GPIO_NUM_36  // default pin: Sig2:4 needs external pullup without input module
 #define CONTROL_CYCLE_START_PIN GPIO_NUM_39  //  default pin: Sig2:3 needs external pullup without input module

/*
```

bdring, 2020-08-26:
There are no defaults. I can add it as an example on the 4x Input page. Very few people use them. Macro buttons are trending though.

I have been getting a lot of practice writing them for people who order pre-programmed ESP32s.. Hopefully a useful style will emerge. I put them here. 

https://github.com/bdring/6-Pack_CNC_Controller/tree/master/machine_defs

closed 2020-09-11 by bdring