bdring/6-Pack_CNC_Controller #29: Feature Suggestion: split VMOT into VMOT+VAUX

StephS, 2020-08-29:
Personally I'd like to be able to run my motors at 36v or 48v and I know there are TMC5160 drivers that support this, but the 6-Pack is limited by the onboard VR and a few other components.
What do you think about splitting VMOT into two rails, one rails specifically for the stepper drivers, another for the rest of board/aux power like fans, etc. This would allow the board to be powered by say, 12v which would allow the cheaper/easier to find fans, and the steppers could be up to 36v without any other changes to the board. It would also eliminate some noise on the AUX, as you could run separate power supplies (a single power supply can still power both).

bdring, 2020-08-29:
These are some good suggestions for even more reasons than you have mentioned. I'll consider it for future revisions. FWIW: The next batch will allow 40V input, due to upgrading the obsolete p/s chip. 

At some point you do have to consider the limitations of current and power dissipation in the Pololu socket.

yngndrw, 2020-08-30:
> At some point you do have to consider the limitations of current and power dissipation in the Pololu socket.
For me, this is most important when using a mixture of onboard and external drivers. E.g. When using JMC servos for the maix axes.

In this case, it would be nice to use a single motor power supply.

Aggebitter, 2020-09-25:
I totally agree on spliting VMOT into VMOT + VAUX. I use ~4A external drivers driven @ 48V on LinuxCNC . Depending on MOSFET used in tmc5160 the six-pack is a replacement candidate on my modified EMCO F1 mill.

fladdie, 2021-12-27:
Is there any news here yet? I am currently looking for a new cnc controller board. I would very much like to run tmc5160 driver with 60V.

StephS, 2021-12-27:
I made a fork of the design here:
https://github.com/StephS/6-Pack_CNC_Controller/tree/Kicad-v1p8s/versions/V1p8s

It includes the split VMOT along with some layout changes.
I've already had 5 boards made, but I need to buy parts and assemble and test.
Gerbers are included, so feel free to download and send to your favorite fab. However I haven't rebuilt the BOM yet.

fladdie, 2021-12-28:
Hi Stephanie,That sounds great. I will order a few boards tonight.Is it possible to run the drivers at 60v with your modified board?In the PDF 6_axis_I2S_v1p8s.pdf vmot is labeled with 12-40v.

https://github.com/StephS/6-Pack_CNC_Controller/tree/Kicad-v1p8s/versions/V1p8s

StephS, 2021-12-28:
The limiting factor will be the capacitors for the stepper drivers, and the stepper drivers themselves.
That's one of the things I need to verify on the BOM.
The other would be all the footprints and make sure there aren't any manufacturing issues. Apparently the voltage regulator is hard to come by.

So just note that while the design passed DRC and ERC, it's still untested.

If you don't mind waiting a day, I'm thinking I can pull the parts off my old 6-Pack board.

StephS, 2021-12-28:
So I took a quick look, and the result is that the stock capacitors are 330uF @ 50v. This is why the 40v max rating. I looked for replacement capacitors that are rated >60v and couldn't find any in the same package size. The closest was 150uF @63v which is a significant drop in capacity.

You could do a through hole cap as there are 270uF @ 80v which would work quite well. However this would require either modifying the board, or bending the leads and making a fixture to hold them and prevent damage.

Modifying the board would be the best route, by adding holes for through hole capacitors. It could cause an issue with the SMT caps sharing the same pads, but maybe that can be worked around.

Skywalkerf34, 2021-12-28:
Think also that Two capacitor in// give sum of the capacitor, this is what we do for our board here because of form factor :  https://github.com/Skywalkerf34/6-Pack_CNC_Controller/tree/master  
With this way we can keep the capacitor under the Pololu socket

fladdie, 2021-12-28:
First of all, thank you for your quick response.
I would like to use the TMC5160 pro drivers from Bigtreetech. https://www.biqu.equipment/de/products/tmc5160-pro-v1-0 because they can be used up to 60V (56V).
If I understand correctly the problem with Stephanie's board are the capacitors. 

Skywalkerf43 suggests to put two of the 63V capacitors suggested by Stephanie in // to get back the capacity that would be necessary.

I also like the board from Skywalker. Especially the fact that I could buy it directly on Tindie. Unfortunately, it will not be so easy to convert the board to 60V, right?

bdring, 2021-12-28:
I don't know the details of that controller, but often the 5V DC-DC power supply can not handle inputs as high as 60V.

I would be careful running those drivers close to 60V. Back EMF from the drivers can causes spikes a few percent above the VMot. I would suggest 10% derating.

StephS, 2021-12-28:
Completely agree. And I checked the biqu website, and while the driver states they support up to 60v, 56v is the recommended max. Looking at the TMC5160 datasheet, I have an idea why.
The VSA pin on the 5160 drives an 11.5v internal voltage regulator. On the datasheet section 3.2 it recommends not driving this more than 40v due to significant self-heating.
These stepstick style drivers aren't the best with heat dissipation either, so you'll likely need to derate your max current.

As for my fork of the board, the layout would need to be significantly changed to support two capacitors.

The fork by @Skywalkerf34 uses a voltage regulator with a max of 50v, so you'd be limited there.

If you really want 60v, the TMC5160 datasheet recommends 100uf of bulk capacitance per amp of drive current.
I can look into making room for a larger capacitor, but it definitely becomes a bit difficult with space and layout constraints.

Skywalkerf34, 2021-12-30:
If you power with 56 volts PSU, There might be a way to fork my design cutting some track, using external fuse  and use the footprint to modify power inlet , then change the 2 SMD capacitor for one 100 mF 63V wired in same place.
Of course this will be a  fork handMade for you....let me know if you can be interested in