bdring/6-Pack_CNC_Controller #31: ESP32 do support 5V input on GPIO's

Aggebitter, 2020-09-13:
I have been using 5 volt devices on ESP32 for years no issue at all and already in 2016 it is confirmed by Espressif.
The ESP32 does actually have an internal snapback circuit that protects it from overvoltage.
The Wiki should address this part in sections that says the opposite. 

From an old tread at Hackaday
https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/

The facebook group on ESP8266 where the confirmation is made by Espressif CEO Mr Teo Swee Ann
https://www.facebook.com/groups/1499045113679103/
search for Teo Swee Ann in the group

Teo Swee Ann 
"i can reply officially here: it is 5V tolerant at the IO. while the supply voltage is at 3.3V."

Teo Swee Ann 
"Baoshi Zhu **ESP32** and ESP8285 are both 5V tolerant as well."

Teo Swee Ann 
"but for ESP32, it is a very complicated matter. it supports 1.8V operations too... i don't know where to start..."

Teo Swee Ann 
 "the reason is too many users took it to mean that the chip is 5 V tolerant. When we say 5 V tolerant, we are only referring to the IOs. So some users mistook this to make that they can power the chip entirely off the 5 V supply. The correct usage is to use 5 V open for these 5 V tolerant pins, and only via **only drain configuration**."

Regards
"Onkel David now with a 6-pack"

bdring, 2020-09-16:
Thanks for the information. I did not know that.

Aggebitter, 2020-09-25:
Closing it
A remark when using mechanical limit switches. The signaling voltage is higher at 5V and when doing analog filtering the filter works a lot better as the SNR gets higher from the stepper driver noise. That is why I always used 5V signaling for my endstops.

closed 2020-09-25 by Aggebitter