bdring/6-Pack_CNC_Controller #33: Possible 3D Printed Case Concept

bdring, 2020-09-24:
I prototyped a bit of a new case concept for the 6 Pack. I wanted to keep the flexible, modular concept. If you change out a module, you shouldn't need to redesign the case or even disassemble it.

This uses a base and a partial cover for the motherboard. There would then be individual covers for each module. The design is far from complete, but I printed a few pieces, to get the feel for it. There would be a mounting flange along the bottom and possibly some mounting features for the module covers.

Any thoughts or ideas are appreciated.

[image]

[image]

[image]

[image]

jaysettle, 2020-09-26:
This looks good, I would like to have the file if it's available.

bdring, 2020-09-26:
It is not finished yet. I will post them soon.

What are your thoughts on mounting holes. I thought a flange could be added to the sides. Another thought is to have mounting holes on the floor of the base. It could keep the size down and look cleaner.

What are your thoughts on module covers. I am OK with no covers.

jaysettle, 2020-09-27:
Mounts on the bottom and sides would be great this can be added. I will use them if they are there.
[image]
[image]

bdring, 2020-09-28:
I don't like the cover mounting. It make it difficult to remove when mounted. I made 2 versions. I can post them both.

Side mount

[image]

Internal mount

[image]

jaysettle, 2020-09-28:
My install might be a one off. I won't be able to use either of those I'm using an ammunition case as a controller home. I can modify the file on my own though no problem.

Bleeptech, 2020-09-28:
Looks like a nice bulletproof case build. ;D 
I'm going for fireproof myself and recycling old fire alarm panel boxes for mine. LOL
[image]
 [image]
They're about the same gauge of steel as the ammo cans and you can usually find them cheap or free in the recycle pile at 
your local alarm installer.

bdring, 2020-09-29:
The files have been added to the Repo.

https://github.com/bdring/6-Pack_CNC_Controller/tree/main/3D_Files/case

Click on the files to use GitHub's 3D viewer.

truglodite, 2021-03-14:
Bart, the case looks decent, however for myself I would have to redesign it so it would not get aluminum chips inside. Instead of doing all that, I made a larger "dust hardened" box to house all of the electronics (6pack, pi3, 7" touchscreen, psu, relays, 80mm fan, 40mm fans per hot device, etc). It's a one off design made from mahogany (looks purdy for sure).

Ideally the 6pack enclosure (any cnc controller enclosure for that matter) should come in one of 2 types, 1) A tinker friendly coverless mount designed to plop into a larger dust hardened box, which has integrated fan cooling, or 2) A stand alone dust hardened enclosure for just the controller board. What you have right now doesn't fit type 2 requirements, and the standard feet type mount you have is a better fit for type 1. IMHO, an enclosure should be designed to meet type 2 requirements... dust filtering for the fan ports at least, and careful design that does well at shedding chips (ie, not place for alum chips to hide and drop inside when opening the enclosure). 

I did some remixing/designing to make a "type 2" enclosure for my old ramps setup. It had a "pitched roof with gutters" on top, and a scotchbrite filter for the fan. So I have some ideas I've already brainstormed for the 6pack. Let me know if you want help on this.

Side note... I RE'd the stl model for the v5 board to design a 1pc open mount for my rig. In doing so, I noticed the hole spacing was a bit smaller than the board itself (~2mm short on the diagonal). Scaling my model to 104% fixed it. Not sure if that's an error on JLC's part, or maybe you nudged the holes out a bit and just forgot to update the stl.