bdring/6-Pack_CNC_Controller #35: PCB manufacturing

simonemery555, 2020-10-26:
Hi Bart,
i live the UK so the postage from tindie is pretty expensive, so i thought i would get it made and have maybe with the minimum orders of 5 usually, id have spare for my other projects.
Please can you let me know where you had them made? did you assemble then your self? i searched on PCBWay but it was not there

thanks in advance!

Simon

bdring, 2020-10-26:
I buy from PCBWay. I did do any order sharing there.

I can ship for $18, but it could take a while. Probably about the same time as a PCB way order. 4-6 weeks.

simonemery555, 2020-10-26:
is the 3-6 days listed on their website not realistic?

bdring, 2020-10-26:
That is for raw boards. Assembled boards take at least 3-4 weeks. This
board has some terminal blocks that are not available in China, so that
adds some time. My last order was placed on 9/17 and arrived 10/24.

simonemery555, 2020-10-26:
ah crap. i've not ordered them before. thanks for the eye opener :)

do you have any ready made to ship?

bdring, 2020-10-26:
Yes, I have a lot of the main boards.

I am out of stock on the 4x input modules to interface switches. I am
tracking a delivery on those for Wednesday. If you order from Tindie on
Thursday I should have those.

simonemery555, 2020-10-26:
okay - do you have a link for the $18 shipment?

bdring, 2020-10-26:
I do not. It will ship USPS First Class Package. If you place an order, I will refund the difference from the higher priced shipping.

 I do not want to add that shipping option, because some countries would be way too slow during this pandemic.

simonemery555, 2020-10-26:
sure, understood. I'll stick the order in.

simonemery555, 2020-10-26:
6 Pack Universal CNC Controller - Include Programmed ESP32: Include Programmed ESP32
External Stepper Motor Driver Adapter x 3
0-10V Spindle CNC I/O Module

Shipping: $55.00

It seems to just add every shipping cost up..

bdring, 2020-10-26:
I will refund to the actual USPS First Class Package Cost.

simonemery555, 2020-10-27:
order number #223992, looking forward to the refund of postage! :P 

thanks!

simonemery555, 2020-10-27:
oh and also my cnc is currently a 3 axis, with a moving bed, but i'm adding a new rotational 4th axis which is why im moving to this setup.
The XYZ axis are external nema 23 motors/drivers, the 4th is just a small nema 17 so i will just be using a small driver on board as the A axis.

bdring, 2020-10-27:
I'll probably ship it tomorrow and refund the balance.

Closing this issue. Will follow-up with email.

closed 2020-10-27 by bdring