bdring/6-Pack_CNC_Controller #36: Laser ON/OFF with CNC Relay Module

jaysettle, 2020-11-04:
I have a small 12V laser I've mounted on my gantry and I want to know if it's possible to utilize the relay module installed on slot 3 of my 6 pack to control the ON/OFF off my laser. I'd like it to work with Gcode that's written with for a laser. I don't care about laser strength in terms of the 0 to 255 scale.

bdring, 2020-11-04:
It is a bit of a hack, but you could use it like a relay spindle. 

Ideally you should use a 5v output module in laser mode.

jaysettle, 2020-11-05:
I'm ok with using the 5V output module. I can come up with some hardware relay circuit to fire the 12V laser with the 5V output module. 

Question
What do I need to add in my machine file to turn on the 5V output(output 1 pin) on module slot #5 so that my gcode generated from Fusion 360 using grbl laser post processor turns it ON/OFF. 

I've made an example Gcode file below. Looks like 'G0 S255 M4' turns it on and ... well, looking at the gcode I'm not sure what turns it off between circles...

```
%
(1001)
G90 G94
G17
G21

(2D Profile1)
G54
G0 S255 M4
G0 X18.6 Y-21.744
G3 X21.4 Y-18.256 Z0 I1.4 J1.744 F200
G3 X18.6 Y-21.744 I-1.4 J-1.744
G0 X17.507 Y-23.433
G3 X22.493 Y-16.567 I2.493 J3.433 F200
G3 X20 Y-15.757 I-2.493 J-3.433
G1 X19.961 Y-15.758
G3 X17.507 Y-23.433 I0.039 J-4.242
G0 X17.298 Y-24.207
G3 X25 Y-20 I2.702 J4.207 F200
G1 Y-19.969
G3 X20 Y-15 I-5 J-0.031
G3 X17.298 Y-24.207 I0 J-5
G0 X15.949 Y-25.965
G3 X20 Y-27.211 I4.051 J5.965 F200
G3 X20 Y-12.789 I0 J7.211
G3 X12.789 Y-20 I0 J-7.211
G1 Y-20.025
G3 X15.949 Y-25.965 I7.211 J0.025
G0 X14.543 Y-27.696
G3 X25.457 Y-12.304 I5.457 J7.696 F200
G3 X20 Y-10.566 I-5.457 J-7.696
G1 X19.97
G3 X10.566 Y-20 I0.03 J-9.434
G3 X14.543 Y-27.696 I9.434 J0
G0 X13.808 Y-29.309
G3 X20 Y-31.18 I6.192 J9.309 F200
G1 X20.035
G3 X31.18 Y-20 I-0.035 J11.18
G3 X8.82 Y-20 I-11.18 J0
G3 X13.808 Y-29.309 I11.18 J0
G0 X12.221 Y-32.55
G3 X27.779 Y-7.45 I7.779 J12.55 F200
G3 X20 Y-5.235 I-7.779 J-12.55
G1 X19.962
G3 X5.235 Y-20 I0.038 J-14.765
G3 X12.221 Y-32.55 I14.765 J0
G0 X11.028 Y-34.474
G3 X20 Y-37.029 I8.972 J14.475 F200
G3 X20 Y-2.971 I0 J17.029
G3 X2.971 Y-20 I0 J-17.029
G1 Y-20.056
G3 X11.028 Y-34.474 I17.029 J0.056
G1 S0
M30
%

```
Should look like this. Just some test circles. This with the laser manually wired on so it stays on. (The laser moves to the next circle so fast it appears to turn it off but it is on all the time)
[image]

jaysettle, 2020-11-09:
> It is a bit of a hack, but you could use it like a relay spindle.
> 
> Ideally you should use a 5v output module in laser mode.

I'm trying to use the 5V module for this now per your suggestion. However I get some compile errors about "Relay" is not a spindle type. I tried "LASER" also. 

My machine definition file...
```
#pragma once

#define MACHINE_NAME            "XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2"

#define N_AXIS 6

// I2S (steppers & other output-only pins)
#define USE_I2S_OUT
#define USE_I2S_STEPS
//#define DEFAULT_STEPPER ST_I2S_STATIC

#define I2S_OUT_BCK      GPIO_NUM_22
#define I2S_OUT_WS       GPIO_NUM_17
#define I2S_OUT_DATA     GPIO_NUM_21

#define X_DISABLE_PIN           I2SO(0)
#define X_DIRECTION_PIN         I2SO(1)
#define X_STEP_PIN              I2SO(2)

#define X2_DIRECTION_PIN        I2SO(4)
#define X2_STEP_PIN             I2SO(5)
#define X2_DISABLE_PIN          I2SO(7)

#define Y_DISABLE_PIN           I2SO(8)
#define Y_DIRECTION_PIN         I2SO(9)
#define Y_STEP_PIN              I2SO(10)

#define Z_DIRECTION_PIN         I2SO(12)
#define Z_STEP_PIN              I2SO(13)
#define Z_DISABLE_PIN           I2SO(15)

#define STEPPER_RESET           GPIO_NUM_19

// Socket #1
#define X_LIMIT_PIN             GPIO_NUM_33
#define Y_LIMIT_PIN             GPIO_NUM_32
#define Z_LIMIT_PIN             GPIO_NUM_35
#define PROBE_PIN               GPIO_NUM_34

// Socket #5
#define SPINDLE_TYPE            SpindleType::RELAY
#define SPINDLE_OUTPUT_PIN      I2SO(24) // No PWM
#define SPINDLE_DIR_PIN         I2SO(25)
#define COOLANT_MIST_PIN        I2SO(26)
#define COOLANT_FLOOD_PIN       I2SO(27)

// === Default settings
#define DEFAULT_STEP_PULSE_MICROSECONDS I2S_OUT_USEC_PER_PULSE
#define DEFAULT_HOMING_SQUARED_AXES bit(X_AXIS)

```

bdring, 2020-11-09:
Here are the spindle types we support. 

```
enum class SpindleType : int8_t {
    NONE = 0,
    PWM,
    RELAY,
    LASER,
    DAC,
    HUANYANG,
    BESC,
    _10V,
    H2A,
};
```

They are case sensitive so it must be...

`SpindleType::RELAY`

I cut and paste the machine definition in your last post and it compiled fine for me.  in both PlatformIO and the Arduino IDE.

jaysettle, 2020-11-11:
I've loaded update grbl esp32 successfully now, but am getting an error when laser mode is enabled and I run the gcode in my post above. I'm not sure what's going on, can you help?
```
Skipping comment-only line: (%)
Skipping comment-only line: (1001)
>>> G90G94
ok
>>> G17
ok
>>> G21
ok
Skipping blank line #6
Skipping comment-only line: (2D Profile1)
>>> G54
ok
>>> G0S255M4
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008f9c1 on core 1
Backtrace: 0x400935e4:0x3ffd2c60 0x40093815:0x3ffd2c80 0x4008f9c1:0x3ffd2ca0 0x400f92c7:0x3ffd2ce0 0x400dc82b:0x3ffd2d00 0x400dc6f9:0x3ffd2d20 0x400dc7b5:0x3ffd2d40 0x400dcce2:0x3ffd2d60 0x400d4025:0x3ffd2d80 0x400d77fd:0x3ffd2e40 0x400d820d:0x3ffd2e60 0x400d452f:0x3ffd2fd0 0x400d2c4f:0x3ffd2ff0 0x400fad09:0x3ffd3010 0x4008fd2d:0x3ffd3030
Rebooting...
[MSG:Grbl_ESP32 Ver 1.3a Date 20201022]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2]
[MSG:Axis count 6]
[MSG:I2S Steps, Stream]
[MSG:Init Motors]
[MSG:X  Axis Standard Stepper Step:I2SO(2) Dir:I2SO(1) Disable:I2SO(0) Limits(-2357.000,0.000)]
[MSG:X2 Axis Standard Stepper Step:I2SO(5) Dir:I2SO(4) Disable:I2SO(7) Limits(-2357.000,0.000)]
[MSG:Y  Axis Standard Stepper Step:I2SO(10) Dir:I2SO(9) Disable:I2SO(8) Limits(-1213.000,0.000)]
[MSG:Z  Axis Standard Stepper Step:I2SO(13) Dir:I2SO(12) Disable:I2SO(15) Limits(-150.000,0.000)]
[MSG:Warning: Spindle output pin I2SO(24) cannot do PWM]
[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin I2SO(27)]
[MSG:Mist coolant on pin I2SO(26)]
[MSG:X  Axis limit switch on pin GPIO(33)]
[MSG:Y  Axis limit switch on pin GPIO(32)]
[MSG:Z  Axis limit switch on pin GPIO(35)]
[MSG:Probe on pin GPIO(34)]
Grbl 1.3a ['$' for help]
>>> G0X18.6Y-21.744
[MSG:'$H'|'$X' to unlock]
[Error] An error was detected while sending 'G0X18.6Y-21.744': (error:9) G-code commands are locked out during alarm or jog state. Streaming has been paused.
**** The communicator has been paused ****

**** Pausing file transfer. ****

```

bdring, 2020-11-11:
I am not sure why it crashed. I cannot repeat the problem on my end. 

You do have a few issues that might contributing to the problem.

[MSG:Warning: Spindle output pin I2SO(24) cannot do PWM]

None of the I2O(xx) pins can do PWM. You can only use $Spindle/Type=Relay with those pins.

It appears you are also trying to use laser mode. That should only be used with $Spindle/Type=Laser (which you cannot use)

Trying to use PWM with a relay will likely damage it. 

Try fixing those 2 problems.

jaysettle, 2020-11-12:
I don’t want to use pwm for this laser, I haven’t looked into that as the what’s available to me currently from the laser is a 12v and a ground so I definitely only need discrete on and off from the 6 pack. 

So is it assumed that I shouldn’t use laser mode with the I2S0(xx) pins, even though I don’t need to pwm the laser? I could move the module to another slot if that gets me the ability for gcode created for a laser to turn it on/off. Or should I just use type::spindle and keep the module where it is? Thanks.

bdring, 2020-11-12:
I  would suggest the following.

- Move the module to slot 3 or 4
- Change to $Spindle/Type=Relay

jaysettle, 2020-11-15:
Good progress. Everything works as expected with laser mode one $32=1.

The laser turned off in accordance to the bold texts below which is perfect.
```
%
(1001)
G90 G94
G17
G21

(2D Profile1)
G54
G0 S255 M4
G0 X18.6 Y-21.744
G3 X21.4 Y-18.256 Z0 I1.4 J1.744 F200
G3 X18.6 Y-21.744 I-1.4 J-1.744
**G0 X17.507 Y-23.433**
G3 X22.493 Y-16.567 I2.493 J3.433 F200
G3 X20 Y-15.757 I-2.493 J-3.433
G1 X19.961 Y-15.758
G3 X17.507 Y-23.433 I0.039 J-4.242
**G0 X17.298 Y-24.207**
G3 X25 Y-20 I2.702 J4.207 F200
G1 Y-19.969
G3 X20 Y-15 I-5 J-0.031
G3 X17.298 Y-24.207 I0 J-5
G0 X15.949 Y-25.965
G3 X20 Y-27.211 I4.051 J5.965 F200
G3 X20 Y-12.789 I0 J7.211
G3 X12.789 Y-20 I0 J-7.211
G1 Y-20.025
G3 X15.949 Y-25.965 I7.211 J0.025
**G0 X14.543 Y-27.696**
G3 X25.457 Y-12.304 I5.457 J7.696 F200
G3 X20 Y-10.566 I-5.457 J-7.696
G1 X19.97
G3 X10.566 Y-20 I0.03 J-9.434
G3 X14.543 Y-27.696 I9.434 J0
**G0 X13.808 Y-29.309**
G3 X20 Y-31.18 I6.192 J9.309 F200
G1 X20.035
G3 X31.18 Y-20 I-0.035 J11.18
G3 X8.82 Y-20 I-11.18 J0
G3 X13.808 Y-29.309 I11.18 J0
**G0 X12.221 Y-32.55**
G3 X27.779 Y-7.45 I7.779 J12.55 F200
G3 X20 Y-5.235 I-7.779 J-12.55
G1 X19.962
G3 X5.235 Y-20 I0.038 J-14.765
G3 X12.221 Y-32.55 I14.765 J0
**G0 X11.028 Y-34.474**
G3 X20 Y-37.029 I8.972 J14.475 F200
G3 X20 Y-2.971 I0 J17.029
G3 X2.971 Y-20 I0 J-17.029
G1 Y-20.056
G3 X11.028 Y-34.474 I17.029 J0.056
G1 S0
**M30**
%
```

However when I disable laser mode $32=0 and go back to cutting wood, the laser turns on and stays on, this makes sense to me I think because I have 
```
$Spindle/Type=Relay
```

Is there something else I can add to the code to programmatically and truly disable the laser during $32=0 laser mode off? If not I'll add a manual override switch (I might feel better about this) so I don't burn the garage down.

bdring, 2020-11-15:
At this time the best way is to add a mechanical switch.

Currently most of the hardware setup is done when you compile the firmware. We are are working on a huge project to allow run time setup of the hardware. This will allow on the fly changes like you want. It will affect virtually every hardware type (switches, spindles, lasers, motors, etc), so it will take some time to finish.

jaysettle, 2020-11-15:
Ok this sounds great. 

Hey do you guys have a donation channel?

bdring, 2020-11-15:
Thanks, 

You can do a [one time donation here](https://github.com/bdring/Grbl_Esp32#donation).

You can also give a few dollars a month via the [sponsor link at the top of this page](https://github.com/bdring/Grbl_Esp32).

closed 2020-11-22 by bdring