bdring/6-Pack_CNC_Controller #37: 6 axis for openPNP

liwajaya, 2020-11-18:
hello mr. Bdring, thank you for the amazing program you created, I hope that if you have spare time you can give examples of pins to use on the [OpenPNP](https://openpnp.org/) board, or maybe someone has used it for openpnp.

Thanks.

bdring, 2020-11-18:
I have no experience using OpenPnP. I have heard several people are building some sort of PnP using the 6 Pack and Grbl_ESP32. Some people might be using their own hardware and Grbl_ESP32.

Here is a [link to our Slack](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA). Maybe you could ask questions there. I don't know if any of the PnP users are monitoring Slack, but it might be a good place to start.

liwajaya, 2020-11-21:
thanks for the answer.  I hope there are users who have used it and are willing to share.  in the meantime I may need to join the community in the openpnp group. Thanks.

closed 2020-11-22 by bdring