bdring/6-Pack_CNC_Controller #38: Pre-sales I/O questions

frdfsnlght, 2020-11-23:
I apologize if this is not the correct place to ask these questions. If not, please direct me.

I'm looking to get one of these nifty controllers for an upcoming MPCNC build. I'm in the design phase and I'm trying to get a handle on wiring needs and such. Consider me fairly proficient when it comes to micro controllers/programming/electrical. The questions:

1. **Spindle/laser**
 Initially, I plan on using an off-the-shelf router with manual speed control. But I'd like to have on/off control using GRBL. Someday, I may add a "real" spindle with a VFD, and a laser. To that end, I envision using a "0-10V Spindle Module". Initially, I imagine hooking up the forward direction output to a relay to control the AC power on the router and ignoring the variable output. Obviously, moving to VFD spindle would be "easy" using this module. I imagine I could use the same forward direction signal as a "laser enable" output and of course the variable voltage to control a CC laser driver (like a BlackBuck 8M). Is this a reasonable approach? Would there be any problems using GRBL's laser mode?
1. **Endstops**
 Normally, dual endstops on an MPCNC are wired in parallel/serial such that only a single input for each axis (X/Y) is needed, usually the "X min" and "Y min" on most controllers. Assuming I'm using a "4x Input Module" for this purpose, does the 6-pack support this, or must each limit switch have a discrete input (2x X min, 2x Y min)? I'm thinking of adding limit switches on the maximum end of each axis too. Can they be wired in a similar manner to only consume a single input per axis? I'd also add an additional input module to handle min/max on the Z axis.
1. **Other inputs**
 Assuming 2 input modules as described above, I'd have 2 spare inputs. Does GRBL support pausing/resuming operation via a switch input? I imagine hooking one of the spare inputs to a switch (momentary or toggle?) and using it to pause the job, maybe to clear or check something, then resume again when pressed/toggled. Can this be done?
1. **Outputs**
 I'd like to add a "5V Output Module" and hook some of it's outputs up to relays. In particular, I'd like to hook an output up to control an air solenoid for air blast and control it with GRBL's "coolant" control. Would this be the correct way to it, or is there a more suitable GRBL control mechanism for this?

Any help would be appreciated including machine configurations, tips, or criticisms.

bdring, 2020-11-24:
In general, I would not recommend too much hacking, like using the forward direction spindle control to control the on/off. I would define everything normally and recompile when you change your system.

1.  Spindle. If you want to use the 0-10V to drive a relay You could use the forward direction circuit as a signal to a relay circuit, but I would define a SPINDLE_OUPUT_PIN to the GPIO pin going to that and use SpindleType::RELAY. You can use VFD later and redefine the pins and spindle type accordingly. I do not recommend using that module for a laser. I would get a 5V output module for that.

2. Switches. You can wire 2 switches to each input. The wiring scheme (parallel/serial) depends on whether you use N.C. or N.O. switches.

3. Other inputs: We support feedhold, resume, reset, door, probe, and 4 macro buttons.

4. Outputs. The 5V module cannot directly drive relays. It can provide a signal to a relay control circuit. The Quad MOSFET can supply the current to directly drive relays and solenoids. We support Mist and Flood as well as 4 digital output and 4 PWM outputs.

frdfsnlght, 2020-11-24:
I'm curious why you wouldn't recommend the 0-10V module to drive a laser? I would think that's a good fit. Is it because most lasers are PWM controlled (in which case, I agree with you and a 5V output module would be better)? How is the analog output derived? I took a quick look at the schematic (emphasis on "quick") and it looks like maybe you're smoothing a PWM signal and running it through an adjustable gain amp. Is that right? If that's true, is there a reason you didn't break out the PWM into another output pin on that module? Seems like a desirable enhancement. Best of both worlds.

bdring, 2020-11-24:
You are correct on the design description.

We are just trying to optimize each module, reduce confusion and keep cost
down. I tried "best of X worlds" for years. I like this system better.

jvanbrecht, 2020-12-01:
I'm looking at replacing my Xpr V4 from Spark Concepts, and I came across this today.  My main issue is currently the pwm output is fluctuating, and my spindle speed goes bonkers on anything other than full speed at 10v.  The other issue is that the spindle requires a 10 to 26V line (not sure why, everything is in German).  If I read things correctly (it's been almost 25 years since my EE degree which I never ended up using anyways), the pwm module only provides an additional 5v.. can this be boosted to 10v (the vendor that sold me the spindle stated that it just needs a voltage between 10 and 26 that is constant, does not matter what it is)