bdring/6-Pack_CNC_Controller #48: Removable power supply module

keax, 2021-05-24:
Hello,

First of all, many thanks for your great job !
I have the V1p5 board version and I have a problem to find the AP64350SP-13 component (it's out of stock or with expensive shipping cost) and I don't find any good replacement. In the V1p8, you use the TPS54360DDAR instead, which looks great...but once again, it's out of stock. I think I found a valid replacement : TPS54560DDAR.
So for now, as AP64350SP-13 is not replaceable by TPS54560DDAR, I can order the V1p8 or create a dirty specific power supply module I will plug on FAN connector (with VMot and 5V) to supply 5V from VMot.

But maybe the best answer is to convert the power supply part to a removable (and upgradable) power supply module ?

bdring, 2021-05-24:
The current electronics shortage makes a lot of parts hard to get.

I have used plug in power supplies in the past.

Dan018, 2021-07-03:
I was going to post the exact same comment. The availability of chips has made several items hard to get. I reworked the PCB to accept one of these as the power supply.

https://m.aliexpress.com/item/4000938525410.html?browser_id=705a5f257db44446a8cdb579a9884a13&aff_trace_key=&aff_platform=msite&m_page_id=oj0af5d7vjocav7a17a6d60c21d24812098dd3638e&gclid=&_imgsrc_=ae01.alicdn.com%2Fkf%2FH4582188eb1634e5bbd59ced55ff12f728.jpg_640x640Q90.jpg_.webp

It would be easy enough to create a module that uses the same pinout but uses whatever components you like (or are available). Imo it would also provide some futureproofing for when components reach EOL.

StephS, 2021-12-28:
It would be easy enough to do both. Have the footprint for the onboard supply, but a header to use a module.
Pololu sells high quality modules. But their pinout is not consistent between models.
https://www.pololu.com/product/2858
I think 5 pin, Vout, GND, Vin, En, PowerGood covers all of the use cases. Then the user can adapt the pins as needed.
Due to the different sizes and pinouts, a mounting hole would be difficult, but a protoboard intermediate might be an option.