bdring/6-Pack_CNC_Controller #49: Quad_Basic_Input_v1p1

trafoman2004, 2021-05-27:
About the Quad_Basic_Input_v1p1 is it outputting 5V directly to the ESP32 inputs, is that right?

truglodite, 2021-06-12:
Yes I observed the same problem with the schematic. I mentioned it a while back in discord, but just heard crickets afterwards. I think the gist is most ppl are buying the opto isolated boards, and folks making DIY input boards probably make up the majority of users who are stumbling upon this particular problem. Nonetheless, it should be fixed.

I used my cnc to cut my own expansion boards for my 6-pack. So I was able to revise the circuit with an added 3v3 LDO to power the chip. I had a bunch of spare AMS1117-3.3 on hand so just used those. The original schematic and board files could easily be updated to use any old LDO, but more caps would also be required.

Since my use required 3x quad input boards, I ended up using an LDO on each board for convenience. It would have been easy to add jumpers and use one LDO to power all 3 input chips, but I chose not to as that makes it too wirey for me. With this in mind, I was wondering if the 6-pack could be revised to have 3v3 on the mainboard, and change the expansion pinouts to use this 3v3 in case it's needed. For example, the p12-p15 pins exist on all the expansion headers, but from what I can see there doesn't seem to be any available module that makes use of them. I figure that may be for dev or future use, but if not maybe one of those could be used for 3v3?

StephS, 2021-12-28:
In this case it's likely better to have the 3.3v voltage regulator on the expansion board, because the 3.3v is being generated on the ESP32 board and not the 6-Pack control board. The supply from the ESP32 varies in current rating, but to isolate noise on the MCU it's best it uses a different supply. A small size low cost linear regulator could be used.