bdring/6-Pack_CNC_Controller #50: Can't control BESC spindle.

Iron-Dan, 2021-07-30:
Hello everybody,

I wanted to drive a brushless motor with the 6-Pack CNC Controller and installed RC Servo/BESC CNC I/O Module on socket #4.
I programmed the BESC with a servo tester and then 
wanted to drive it with the controller, but it didn't work.
The motor was only beeping and not turning, so I compared the PWM signals of the servo tester and the RC Servo/BESC CNC I/O Module with the scope. I found out that GPIO14 delivered the correct pulse widths at commands S0 (1ms) and S1000 (2ms), but the voltage level is about factor 10 too low. The voltage level of the pulses at GPIO14 are just 0.56V, at the servo tester they are 5V.

[image]
[image]
[image]
[image]

Does anybody know why the ESP32 delivers such a low voltage, shouldn't it be 3.3V?
How can I fix this?

Thanks and best regards,
Dan

bdring, 2021-07-30:
Have you tried measuring with with no servo attached? Can you try it with a different pin.

You could have a damaged or overloaded pin.

Iron-Dan, 2021-07-31:
Hi bdring,Thank you for the fast Response.I checked it with another ESP32 and I could control a SG90 servo. My scope at home doesn't work, I will check the signal at work on monday. The BESC is also at work, so I can't check it now. ☹️But it seems like the other ESP has a defective port.Best regards,Dan-- Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

Iron-Dan, 2021-08-01:
Hi bdring,

I got my scope at home running and now I can confirm that everything seems ok at the second ESP32, the voltage level is now 3.3V.

Command S0:
[image]

Command S1000:
[image]

I have another question about the spindle buttons in the WebUI.
Do they only work when a job is running or should I be able to turn the spindle on and off at any time?
I'm wondering because the mist button is working at any time.

At last I would say thank you for the good job at ESP32_GRBL and 6-Pack CNC Controller, they are really cool.
I have sent you a 20$ donation via Paypal.

Thanks and best regards,
Dan