bdring/6-Pack_CNC_Controller #52: Gang Stepper Motor Configuration

Mamoonpk, 2022-04-01:
Hi. 
I have my 6 pack board with me and very exciting to go live with it. however, I am struggling with some settings as I have little knowledge in this area. but I am keen to learn and move forward. 

I have gone through settings multiple times to see how to gang stepper motors. In my case, I need to gang 2 motors for the y-axis. 

further, i am using NEMA 23 with External driver DM556T. I hope DM556T is compatible with 6 Pack. 

please advise.

Mamoonpk, 2022-04-01:
Thanks - I found it [here](https://github.com/bdring/6-Pack_CNC_Controller/wiki/Setting-Up-Axis-Motors)
please let me know If DM556T is compatible with 6 pack board.

nicolas-lang, 2023-11-19:
> Thanks - I found it [here](https://github.com/bdring/6-Pack_CNC_Controller/wiki/Setting-Up-Axis-Motors) please let me know If DM556T is compatible with 6 pack board.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/Using-External-Stepper-Drivers