bdring/6-Pack_CNC_Controller #54: SD Card support?

sshahriyari, 2022-07-02:
Hi,

how I can connect SD Card to my 6-Pack_CNC_Controller ?

plz help me, thanks

bdring, 2022-07-02:
There is an SD socket on the controller.

See this for [FluidNC](https://github.com/bdring/FluidNC/wiki/SD-Card)

sinantasci, 2024-03-24:
Hey Barton,
I bought the 6 pack controller V1.1 http://wiki.fluidnc.com/en/hardware/official/6_Pack_External from Electrolux.
However it fails during the NC program upload. It seems all settings are correct. 
The error code is [MSG:ERR: sdmmc_card_init failed code 0x107] . How can I fix it? I assume there is no problem with the control card.
Many Thanks
Sinan

bdring, 2024-03-24:
Electrolux?

It is likely the format. Everything we know about SD carts is here.

http://wiki.fluidnc.com/en/config/sd_card

sinantasci, 2024-03-24:
Sorry, It was Elecrow. Despite trying numerous troubleshooting steps. 
The card, which is 64GB, has been formatted with FAT32, and I've attempted different volume sizes including 32GB, 4GB, and 2GB. However, each time I receive the error message '[MSG:ERR: mount_to_vfs failed code 0xffffffff].' Have you encountered this issue before?

sinantasci, 2024-03-24:
Addition to this, when I reduce the communication frequency to 400000Hz. I recieve this alarm message. [MSG:ERR: sdmmc_card_init failed code 0x107]. I can create folders in the SD card but can't upload anything.
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 400000

bdring, 2024-03-25:
Start by adding a few files via your pc then reading them using FluidTerm using $sd/list

sinantasci, 2024-03-25:
Adding files via PC helped me to list and run the programs. It seems the only problem I'm encountering is uploading files through the FluidNC user interface.

bdring, 2024-03-26:
Thanks for the info. I will try to repeat the problem as soon as I can. I am driving home for the next 2 days.

sinantasci, 2024-04-05:
Hey Barton, did you have a chance to repeat the issue? 
Thank you