bdring/6-Pack_CNC_Controller #61: how to properly operate relay switch and enable them

erikcramerjr, 2022-12-17:
hi,
ive used install-fs from version fluidnc-v3.6.5-win64 to install the firmware and added the following config :

[config.txt](https://github.com/bdring/6-Pack_CNC_Controller/files/10251582/config.txt)
To enable the relays ive placed them on module 3 and 4, i intend to use one for the spindle and one for the vacuum.
In the near future i might use one also for compressed air.
and and a lcd and opto for the sensors 

to enable the relays i have Unlocked the machine (clear alarm)
Then i pressed the spindle, flood , mist buttons but these dont do any thing.
So i thought i must use the device M6 T0 or T1 
So i noticed that under tool_num  both devices have numer 100 so i changed these to 0 and 1 and tested again, with the same result. (i expect a click from the relay and the LEd to light up)

Do i need to compile the firmware to get this working correctly 

Or do i have to use some thing like this with the relay for the vacuum  Because i thested this but it also didnt work?
 coolant:
  flood_pin: gpio.14
  mist_pin: NO_PIN
  delay_ms: 0

Verbose log from enabling the pins
$G
[MSG:VRB: Cannot open file /spiffs/macrocfg.json]
[MSG:VRB: Cannot open file /spiffs/macrocfg.json.gz]
[MSG:VRB: Cannot open file /spiffs/404.htm]
[MSG:VRB: Cannot open file /spiffs/404.htm.gz]
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|WCO:0.000,0.000,0.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|Ov:100,100,100>
$X
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
[MSG:INFO: Caution: Unlocked]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
m7
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
ok
m8
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
m6 T0
[MSG:INFO: Tool No: 0]
[MSG:INFO: Using spindle Relay]
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZ>
M6 T1
[MSG:INFO: Tool No: 1]
[MSG:INFO: Using spindle Relay]
ok
Image of relay settings, (when i have both relay settings in there like in the config they both show up., this is a screenshot with only one and the other settings was with Coolant option)
[image]
For coolant i can only find this one setting:
[image]

bdring, 2022-12-17:
The buttons on the Webui are overrides. Send M7, M8 and M9 from the console.

erikcramerjr, 2022-12-17:
Thank you Module 4 switches like a Charm