bdring/6-Pack_CNC_Controller #64: tmc2226

ViktorRindom, 2023-01-10:
I could not find any documentation for the tmc 2226 stepper driver. I would like to know how I should set the jumpers and how I should set the axes up using that board in the code.

ViktorRindom, 2023-01-10:
Its most likely like the tmc2209 (so like stepstick) but it would be nice to have that confirmation.

bdring, 2023-01-10:
That is a UART controller stepper driver. The 6 pack does not support those for multiple reasons. Read here on what drivers are supported.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/Supported-Stepper-Drivers

ViktorRindom, 2023-01-12:
i get that i cant run it using uart but can it be run in "step stick" mode?

i already have this driver and would like confirmation that it works like the 2209:
UART type motors like TMC2209 are not supported yet. The TMC2209 can be used in StepStick mode, called SilentStepStick. The UART is not used, so you are forced into 16x microstepping and SpreadCycle (not silent) mode. Set jumper like StepStick drivers.