bdring/6-Pack_CNC_Controller #70: Just an Observation with Probe definition and gpio

Bill83046, 2023-11-24:
In configuration files for probe: I find that it does not conform to other definitions of gpios. 
probe:
  pin: gpio.25
examples: 
      limit_neg_pin:  gpio
      limit_pos_pin:  gpio
      limit_all_pin:  
I would expect it to be as below to follow other gpio definitions.
probe:
probe_pin: gpio
Do not know if this was intentional or just an overlook. Am I reading this correctly?
Previous iterations have probe_pin: gpio
Bill Roberts

bdring, 2023-11-24:
I don't follow your question. Here is the wiki page.

http://wiki.fluidnc.com/en/config/probe

Here is an example

```yaml
probe:
  pin: gpio.34
  toolsetter_pin: gpio.14
  check_mode_start: true
  ```
  
  Initially there was only one pin. That is why the 2 pins look a little different.

Bill83046, 2023-11-24:
All of your gpio definitions have mostly the same format, i.e. name_pin: gpio#. The probe: gpio# definition strays from that format ( a  housekeeping issue with me, I just like to see things defined in the same manner). Just an observation on my part. I know it probably does not make a difference.  I am trying to track down an issue with a probe error not being defined when I upgraded to v3.70.10. I Reverted back to V3.7.9 and is ok. Do not know if it is me yet? When I get to installing 3.70.1 again I will keep track of messages and errors and if necessary, update with issue.

bdring, 2023-11-25:
There are a few items that broke our naming convention. Many things and sections evolved over time. There are many things we would love to change, but breaking thousands of working config files would be a support nightmare.

Bill83046, 2023-11-28:
I get it. I was a controls engineer for Jet Engine Component test stands for 23 years. The controls software that I used and developed was for only one person, not the many that you try to make happy. But you understand where I am coming from. I tend to ask a lot of questions if I do not understand or if you will, question the methods. 
Now running V3.70.1 . I get 2 messages  [MSG:DBG: /macrocfg.json not found] and [MSG:DBG: 404.htm not found]
I assume this has to do with the WEB EUI. Do not know why but it dose not effect the operation so far. Cannot find any details on this on the WEB. 
So far so good my machine is up and running except for VFD Spindleand that is next.  All axises are calibrated accurate and repeatable.
Thanks 
Bill Roberts

bdring, 2023-11-28:
If you have questions about FluidNC, it is better to ask on the FluidNC github repo. More people will see it there.

We also have a Discord server with a WebUI channel.

https://discord.gg/az5DaQS9