bdring/6-Pack_CNC_Controller #73: Esp32 µC getting fried by the 6pack board.

crasu, 2025-05-16:
I have used my 6pack v1p8 for a couple of years now with my mpcnc. I have got a display and to 4xInput modules.

Recently the 6pack started eating µCs. Today I got really bad an I had to replace the esp32 on the board two times and about 2-3 times in the last year. In all cases the voltage regulator on the esp32 boards are fried. I have tried different esp32 boards. I checked the power supply, pin for the esp32 - They are at 5V. I use wifi and have no usb cable connected to the board. 

Are there any known issues with the board that could cause this? Any ideas?

bdring, 2025-05-16:
It is usually due to pulling too much 3.3v from the esp32 module. That heats up the regulator.

crasu, 2025-05-17:
Thank your for your quick response! Any idea how to debug this? I already checked the esp32 with an ohm meter for shorts against ground (nothing). Last year I swapped all stepper drivers (drv8825).